stm32-rs/bxcan #22: version conflicts

pdgilbert, 2021-04-01:
I am building some things that work with multiple device crates and have just started getting `cargo update` errors because one crate has update `bxcan` while another has not:

```
error: failed to select a version for `defmt`.
    ... required by package `bxcan v0.5.0`
    ... which is depended on by `stm32f7xx-hal v0.2.0 (https://github.com/stm32-rs/stm32f7xx-hal#c0bb1f2b)`
    ... which is depended on by `eg_stm_hal v0.1.0 (/home/paul/githubClones/eg_stm_hal/boards/none-stm32f722)`
versions that meet the requirements `^0.2.0` are: 0.2.1, 0.2.0

the package `defmt` links to the native library `defmt`, but it conflicts with a previous package which links to `defmt` as well:
package `defmt v0.1.3`
    ... which is depended on by `bxcan v0.4.0`
    ... which is depended on by `stm32f0xx-hal v0.17.1 (https://github.com/stm32-rs/stm32f0xx-hal#e9c99650)`
    ... which is depended on by `eg_stm_hal v0.1.0 (/home/paul/githubClones/eg_stm_hal/boards/none-stm32f722)`

failed to select a version for `defmt` which could resolve this conflict

```
It seems that `bxcan` demands a specific version of `defmt` and also in the `README.md` usage suggests demanding a specific version of `bxcan` (now `bxcan = "0.5.0"` but I see it is updated automatically).

Any chance you can be more flexible by depending on `defmt = "^..."`  or `">=..."` and suggest similar flexibility regarding `bxcan` in your `README`?

jonas-schievink, 2021-04-01:
Yeah, it would be good to improve this. I think it's mostly up to defmt and Cargo to provide better handling of dependencies that use `links` here, see https://github.com/knurling-rs/defmt/issues/426 and https://github.com/rust-lang/cargo/issues/7880.

pdgilbert, 2021-04-01:
Ok,  I see the `defmt` issue is a bit messy.  But I think crates using `bxcan` can help by specifying `bxcan = ">=0.4" ` rather than `bxcan = "=0.5.0"  ?   Could you at least change the README.md?

jonas-schievink, 2021-06-19:
I've opened https://github.com/stm32-rs/bxcan/pull/35 to make defmt support opt-in, and to allow patch versions of bxcan to update defmt (which normally would be considered a breaking change). This should lift some of the burden from HALs, since they no longer need to update to a new breaking bxcan release every time a new defmt version comes out.

Naturally, the drawback is that the defmt version may change, so that means that mostly only applications (which check in their lockfile) would want to use the defmt feature.