stm32-rs/bxcan #28: Implement `free()` method

Sh3Rm4n, 2021-05-05:
In the case I understood the API correctly, it is currently not possible to retrieve the peripheral consumed with [`Can::new`](https://docs.rs/bxcan/0.5.0/bxcan/struct.Can.html#method.new) associated with the `Instance` trait 

https://github.com/stm32-rs/bxcan/blob/f4d5d9676b1653eaf7e51beb222e5af72e6baf1a/src/lib.rs#L49

To retrieve the peripheral, maybe a `free()` method should be implemented like that:

```diff
diff --git a/src/lib.rs b/src/lib.rs
index 2584107..f589e00 100644
--- a/src/lib.rs
+++ b/src/lib.rs
@@ -470,6 +470,10 @@ where
     pub fn split(self) -> (Tx<I>, Rx<I>) {
         unsafe { (Tx::conjure(), Rx::conjure()) }
     }
+
+    pub fn free(self) ->  I {
+        self.instance
+    }
 }
 
 impl<I: FilterOwner> Can<I> {
```

This is probably not enough, I guess. I have only glanced over the implementation yet, but I imagine, that the peripheral has to be deconfigured to safely free it.

jonas-schievink, 2021-05-05:
Sounds reasonable to have this, yeah. I'm not entirely sure what `free` should do exactly, since some stuff is up to the HAL (like turning off the clock).

Sh3Rm4n, 2021-05-05:
Because `free` in the HAL is usually about freeing the consumed pins, it has to wrap this free function anyways, so leave it up entirely to the HAL?

jonas-schievink, 2021-05-05:
Yeah, that could work

closed 2021-05-15 by jonas-schievink